projectacrn/acrn-hypervisor #8639: Sourcing the Root Cause of "Assertion failed in file acpi_parser/dmar_parse.c,line"

xuyizhe1999, 2024-06-28:
**Bug Description**
I was trying to run ACRN v3.0 on QEMU with the guide:  [Advanced Features > Enable ACRN Over QEMU/KVM](https://projectacrn.github.io/3.0/tutorials/acrn_on_qemu.html).
However, I met the problem of "Assertion failed in file acpi_parser/dmar_parse.c,line" identically as the one mentioned in #8494 and #8571.
Here is the debug outputs.
```
Loading ACRN hypervisor ...
error: Can't get controller info..
Non-UEFI environment detected.
Multiboot Bootloader: GRUB 2.04-1ubuntu26.17
ACRN Hypervisor
[12902244us][cpu=0][(null)][sev=2][seq=1]:HV version 3.0-2024-06-27 08:57:56-506c87cfa DBG (daily tag:acrn-2022w27.1-180000p) shared@qemu build by xu, start time 12891408us
[12903582us][cpu=0][(null)][sev=2][seq=2]:API version 1.0
[12904031us][cpu=0][(null)][sev=2][seq=3]:Detect processor: Intel Atom Processor (Denverton)
[12904745us][cpu=0][(null)][sev=1][seq=4]:SECURITY WARNING!!!!!!
[12905302us][cpu=0][(null)][sev=1][seq=5]:Please apply the latest CPU uCode patch!
[12908019us][cpu=0][(null)][sev=2][seq=6]:Assertion failed in file acpi_parser/dmar_parse.c,line 120 : fatal error
[12909007us][cpu=0][(null)][sev=2][seq=7]:
 delta = (actual_load_address - CONFIG_HV_RAM_START) = 0x0

[12909913us][cpu=0][(null)][sev=2][seq=8]:
Host Stack: CPU_ID = 0

[12910513us][cpu=0][(null)][sev=2][seq=9]:addr(0x1120bb90)      0x0000000000000000  0x0000000000000013  0x0000000011247f00  0x000000007ffe29bc

[12911708us][cpu=0][(null)][sev=2][seq=10]:addr(0x1120bbb0)     0x000000001120bc20  0x0000000011012273  0x0000000011055940  0x000000007ffe2a64

[12912895us][cpu=0][(null)][sev=2][seq=11]:addr(0x1120bbd0)     0xffffffff1120bc10  0x0000000011247f80  0x00000000000000fe  0x0000000000000000

[12914099us][cpu=0][(null)][sev=2][seq=12]:addr(0x1120bbf0)     0x00000000000000fc  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12915308us][cpu=0][(null)][sev=2][seq=13]:addr(0x1120bc10)     0x0000000000000000  0x0000000000000000  0x000000001120bc30  0x000000001100d860

[12916554us][cpu=0][(null)][sev=2][seq=14]:addr(0x1120bc30)     0x000000001120bc60  0x0000000011009499  0x0000000000000000  0x0000000000000000

[12917762us][cpu=0][(null)][sev=2][seq=15]:addr(0x1120bc50)     0x0000000000000000  0x0000000000000000  0x000000001120bc80  0x00000000110081e9

[12918963us][cpu=0][(null)][sev=2][seq=16]:addr(0x1120bc70)     0x00000000110010e2  0x0000000000000000  0x00000000110623d0  0x00000000696e746c

[12920135us][cpu=0][(null)][sev=2][seq=17]:addr(0x1120bc90)     0x0000000000000000  0x0000000000000000  0x0000000100000000  0x0000000000000000

[12921292us][cpu=0][(null)][sev=2][seq=18]:addr(0x1120bcb0)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12922419us][cpu=0][(null)][sev=2][seq=19]:addr(0x1120bcd0)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12923594us][cpu=0][(null)][sev=2][seq=20]:addr(0x1120bcf0)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12924768us][cpu=0][(null)][sev=2][seq=21]:addr(0x1120bd10)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12925951us][cpu=0][(null)][sev=2][seq=22]:addr(0x1120bd30)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12927122us][cpu=0][(null)][sev=2][seq=23]:addr(0x1120bd50)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12928295us][cpu=0][(null)][sev=2][seq=24]:addr(0x1120bd70)     0x0000000000000000  0x0000000000000000  0x0000000000000000  0x0000000000000000

[12929447us][cpu=0][(null)][sev=2][seq=25]:

[12929838us][cpu=0][(null)][sev=2][seq=26]:Host Call Trace:

[12930354us][cpu=0][(null)][sev=2][seq=27]:----> 0x0000000011012273

[12930931us][cpu=0][(null)][sev=2][seq=28]:----> 0x000000001100d860

[12931510us][cpu=0][(null)][sev=2][seq=29]:----> 0x0000000011009499

[12932087us][cpu=0][(null)][sev=2][seq=30]:----> 0x00000000110081e9

[12932664us][cpu=0][(null)][sev=2][seq=31]:

```
**Platform**
Host server contains two Intel(R) Xeon(R) Gold 5218 CPUs, and runs Linux kernel 6.2.0.
Guest configurations are the same as described in the guide.
Interestingly, one of our colleagues met the same problem, but later he was able to successfully run the ACRNv3.0-on-QEMU case on a Intel(R) Core(TM) i5-7500 CPU platformed PC.

**Possible Cause**
We revisited the [ACRN Supported Hardware](https://projectacrn.github.io/latest/reference/hardware.html) and guessed it to be the probable cause of this problem.
Apparently, if we want to run ACRN (with/without QEMU), it'll be better to use the machines and CPUs described in the hardware guide.

However, we still have two questions here:
1. **Is it still possible to run ACRN on a server processor?** As we surveyed, the most possible solution would be Xeon E 2460+ CPU in Raptor Lake family. But it is not officially tested. Besides, Raptor Lake CPUs only support ACRN after v3.2, while ACRN-on-QEMU case is tested up to v3.0.
2. **Is there a supported machine with multiple CPU sockets?** As we also surveyed, it seems impossible to run ACRN on a multiple-socket CPU machine, simply because there is currently no such a motherboard (e.g., with multiple LGA1700 sockets). All supported PCs and Mobiles are single-CPU-socket  machine as well.

jiaqingz-intel, 2024-07-23:
Increasing `MAX_DRHD_DEVSCOPES` in hypervisor/include/arch/x86/asm/vtd.h is all you need

xuyizhe1999, 2024-07-23:
it works when i double the value of MAX_DRHD_DEVSCOPES to 32, thanks!!!

closed 2024-07-23 by xuyizhe1999