projectacrn/acrn-hypervisor #8667: Options of ACRN hypervisor may miss in grub menu after first installation and reboot.

jiayuan-intel, 2024-07-25:
**Describe the bug**
When firstly installed ACRN and boot up a VM, the options related to ACRN may miss after rebooting SOS. The acrn.bin and something else is existed in /boot, and run update-grub will fix. But why the options in the menu will miss once? It's not 100% appeared but both showed on Vecow and Asus mini PC once.

**Platform**
Both showed once on Vecow and Asus mini PC

**Codebase**
release_3.3

**Scenario**
Shared scenario described in Getting Started Guide.

**To Reproduce**
Steps to reproduce the behavior:
1.Follow GSG, finish all the steps and poweroff the VM.
2.Reboot SOS, check grub menu.

[image]
[image]

jiaqingz-intel, 2024-07-25:
```
update-initramfs: deferring update (trigger activated)
Sourcing file `/etc/default/grub'
Sourcing file `/etc/default/grub.d/acrn.cfg'
WARNING: GRUB_DEFAULT changed to boot into ACRN by default!
         Edit /etc/default/grub.d/acrn.cfg to avoid this warning.
Generating grub configuration file ...
Found linux image: /boot/vmlinuz-6.8.0-31-generic
Found initrd image: /boot/initrd.img-6.8.0-31-generic
Found linux image: /boot/vmlinuz-6.1.80-acrn-service-vm-375511-g2355d104a177
Found initrd image: /boot/initrd.img-6.1.80-acrn-service-vm-375511-g2355d104a177
Found linux image: /boot/vmlinuz-6.8.0-31-generic
Found initrd image: /boot/initrd.img-6.8.0-31-generic
Found linux image: /boot/vmlinuz-6.1.80-acrn-service-vm-375511-g2355d104a177
Found initrd image: /boot/initrd.img-6.1.80-acrn-service-vm-375511-g2355d104a177
Found memtest86+ 64bit EFI image: /boot/memtest86+x64.efi
debconf: DbDriver "config": /var/cache/debconf/config.dat is locked by another process: Resource temporarily unavailable
ACRN: Cannot access debconf acrn-hypervisor/board:
```

Failed to read parameters when configuring package

jiayuan-intel, 2024-08-26:
Fixed. Thanks @jiaqingz-intel. 

Close this issue.

closed 2024-08-26 by jiayuan-intel