projectacrn/acrn-hypervisor #8687: Unable to boot on HP Fury 16 G10 Mobile Workstation

mrtrkmn, 2024-08-08:
It is not able to load hypervisor, stucks at loading.

What can be done to debug or fix to solve this issue

mrtrkmn, 2024-08-12:
When I select the shown kernel with service vm, it tries to boot but freezes after "Loading ACRN service VM initial disk ..."

[image]
[image]

jiayuan-intel, 2024-08-26:
Hi @mrtrkmn , it's recommend to use ACRN kernel to boot into Service OS (listed as with Linux 6.1.80-acrn-service-vm-375513-g7159ad071be8(ACRN 3.2) in your screenshot). Your 6.5 kernel may miss some key kernel configs like CONFIG_ACRN_HSM.

mrtrkmn, 2024-08-26:
Thank you very much for your response @jiayuan-intel 
I actually tried both and service vm was also having the same issue. 

Could not move further from this step, not able to debug it as well

jiayuan-intel, 2024-08-26:
It's better to follow ACRN v3.3 Getting Started Guide (https://projectacrn.github.io/latest/getting-started/getting-started.html) to try again. V3.3 is released two weeks ago, and it's officially support Ubuntu 22.04.4 with kernel6.1 as Service OS and Raptor Lake as your platform(13950HX).

If you want to debug, please refer to https://projectacrn.github.io/latest/develop.html#debug . Debug via serial port and enter acrn-console is one of the way to have some helpful logs.

mrtrkmn, 2024-08-30:
I re-tried several times but it does not work at all :( Nevertheless, thank you for your comments @jiayuan-intel